jrsteensen/OpenHornet #709: ECS, BLEED AIR MECHNAMISM - Incorrect nut size

Exprezz0, 2023-04-25:
### Summary:
The size of the jam nut used on the pull shaft coupler for solenoid is incorrectly sized.  

### Expected Results/How It Should Work:
Hole size on pull shaft coupler is for 2-56 x 1.00in screw, which is the screw used and labeled in the model.  The current jam nut is 6-32.

### Screenshots:
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A5A1-200 - ASSY, BLEED AIR KNOB MECHANISM
OH5A5A1-1 - ASSY, ECS PANEL _(guilty by association)_

jrsteensen, 2023-04-25:
@Lazarus94 Can you take care of this as part of the ECS Panel work.

Exprezz0, 2023-04-25:
The drawing for the Bleed Air Knob Mechanism #705 is in-progress by @Lazarus94.  I'll update the model in Fusion and close this, but he'll need to update it when he does the drawing (which hasnt been started yet by the looks of it.)

Lazarus94, 2023-04-25:
@Exprezz0 Cool thank you. Will update in the drawing.

Exprezz0, 2023-04-25:
[image]

closed 2023-04-25 by Exprezz0