jrsteensen/OpenHornet #713: FCS Panel - Interconnect Diagram for Rudder Trim Motorized Pot Needs Updating

Arribe, 2023-05-01:
### Summary:
The L298N driver needs at least to pin connections, with a 3rd optional PWM connection.  The current diagram only shows the PWM connection.

### Expected Results/How It Should Work:
We only need to run one motor, so this assumes connecting to the A side of the driver. The L298N driver uses In1 and In2 to drive the motor direction.  When they are opposite values HIGH/LOW the motor will turn, reverse the values to reverse the motor.  When both pins are the same HIGH or LOW then the motor stops.  The optional PWM value into EnA controls the motor speed.  If we want to motor to turn at top speed, then leave the jumper in place for sending the full 5V.

### Actual Results/How It Does Work:
The interconnect diagram only shows the PWM connection to the servo and omits the L298N driver.  The diagram should include both the driver and the wiring to the motor for clarity.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
Interconnect Diagram for 4A6A1 ABSIS_ALE

jrsteensen, 2023-05-01:
Complete on #389

jrsteensen, 2023-05-08:
Will be completed on #389

closed 2023-05-08 by jrsteensen