jrsteensen/OpenHornet #714: OH4A6A1: Rudder Trim Mechanism Needs Reduced Internal Friction.

Arribe, 2023-05-01:
### Summary:
The 6389k231 flanged sleeve bearings add too much friction to the gear preventing the motorized pot from turning.

### Expected Results/How It Should Work:
The motor should be able to turn the pot without too much resistance.  

### Actual Results/How It Does Work:
When barely tightening the mechanism the flanges facing the gear clamp down on it preventing the motor from turning.  Perhaps replace with a sleeve that is not flanged or space out top and bottom so that the gear can be held in place in the middle with a gap.  The rod can ride on the lower metal 6338K561 bearing and does not need to be supported by the gear.

### Screenshots:
Here is the tightest it can go while still turning.

[image]

This shows that the top and bottom plastic bearing make a clamp for the gear when the parts are tightened.
[image]

A potential approach, replace the 6389k231 with a centering bushing, or shrink the top and bottom printed parts to not need a sleeve.  Then the gear can ride freely in the middle held in place by the set screw.
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [x] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH4A6A1: Rudder Trim Mechanism

jrsteensen, 2023-05-07:
@Arribe  I'm a little bit confused since the mechanism shows the top flanged bushing being recessed which leaves a small clearance gap for the gear, so it should not bind.

[image]

jrsteensen, 2023-05-07:
I'm not seeing that same recess on yours, though it could just be the angle or the slightly low resolution image.

jrsteensen, 2023-05-07:
Previously resolved in the model when we flipped the gears.

closed 2023-05-07 by jrsteensen