jrsteensen/OpenHornet #716: Wing Fold Mechanism Handle Missing Retaining Screw

Lazarus94, 2023-05-03:
### Summary:
There is no screw to hold the Wing Fold handle onto the main shaft. 

### Expected Results/How It Should Work:
A screw should go through the modelled holes to hold the handle onto the shaft. The shaft also needs to be rotated in the CAD model.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A2A2-1 - ASSY, WING FOLD HANDLE

jrsteensen, 2023-05-03:
Verify it's not just a hidden body keeping it from being visible before adding the screw.

jrsteensen, 2023-05-05:
Fixed in Wing Fold Handle v13

closed 2023-05-05 by jrsteensen