jrsteensen/OpenHornet #719: ECAD Directory Overhaul - Part 2

jrsteensen, 2023-05-05:
Tactical Pause.  Directory overhaul is nearing completion with just a few things left to do:

- [ ] ABSIS Transition to KiCad 7  (est: ~3hrs)
   - [ ] Change path of Drawing Sheet for schematic and PCB
   - [ ] Verify paths of symbols/footprints within schematic
   - [ ] Final review of manufacturing files and assets
- [ ] Backlight Panels - Type A (est: 2-3 days) - going alphabetically
   - [x] ANT SEL thru COMM 
   - [ ] DEFOG through SPIN RCV 
- [ ] Functional PCBs Transition to KiCad 7 (est: ~8-10hrs)
   - [ ] Change path of Drawing Sheet for schematic and PCB
   - [ ] Verify paths of symbols/footprints within schematic
   - [ ] Final review of manufacturing files and assets
- [ ] Fix any outstanding issues
- [ ] Final Review of Directory:   Check directory and ensure there are no duplicate, excess, or corrupted files.
- [ ] Update and Review Manufacturing.md and ReadMe.md
   - [ ] Add instructions for alternate parts as needed
- [ ] MasterBOM! (Requires everything else to be complete)

Exprezz0, 2023-05-17:
ABSIS Transition complete.  Could probably use one more review just to make sure everything has updated symbol properties.

Exprezz0, 2023-06-16:
Backlight PCB Transition complete.  That's the bulk of the work.  Just need to go through the functional stuff and double check.  Then MasterBOM stuff!

Exprezz0, 2023-07-25:
Need to decide what to do with the Left Circuit Breaker PCB.  Technically, it's not implemented in the assembly drawing, but was completed and exists to allow for use of backlighting the circuit breaker.  I put it in the release package and named it "OH4A8A1-XX - PCB, LEFT CIRCUIT BREAKER" pending feedback.

jrsteensen, 2024-02-25:
Will be continuing this effort in next release.

jrsteensen, 2024-10-19:
This issue is abandoned. Will continue to implement on an attrition basis as required. Will reopen if work is to be continued on it.

closed 2024-10-19 by jrsteensen