jrsteensen/OpenHornet #725: Replace

jkkicks, 2023-05-13:
Mini Din 60J connector listed in Stick Assembly is the incorrect connector and should be updated to interface with thrustmaster f18 Stick

Should be replaced with:
Digikey: CP-2650-ND
OR
Digikey: CP-2750-ND

The former seems to be discontinued but is similar to the part originally used by thrustmaster in their base

### Screenshots:
[image]

[image]

### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [  ] Center Tub
- [ X ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*OHA3-40*

jrsteensen, 2023-05-13:
- [ ] Evaluate MD-50SV connector: https://www.digikey.com/en/products/detail/cui-devices/MD-50SV
- [ ] Evaluate free hanging equivalents.

- MD-50SP is obsolete and not suitable for new design.

jrsteensen, 2023-05-13:
Ordered a MD-50PL100 for evaluation and fit check: https://www.digikey.com/en/products/detail/cui-devices/MD-50PL100/500828

jrsteensen, 2023-05-20:
Verified MD-500PL100 compatibility.

[image]

Starting revision/integration of design.

jrsteensen, 2023-05-20:
Fixed in stick extension v14 and dwg v2

commit ee666e7a (2023-05-20): Updated OH3A34-40 drawing

Resolved #725
closed 2023-05-20 by jrsteensen