jrsteensen/OpenHornet #726: OH1A8A1-115 - UFC INNER SPACER PLATE

Fedhi216, 2023-05-15:
### Summary:
There are problems with the design of the "INNER SPACER PLATE":

-The holes for the "OH1A8A1-101 - UFC COMM BEZEL" brackets are not present on the left side.

-Impossible to laser cut the holes for the "OH1A8A1-101 - UFC COMM BEZEL" brackets as shown in the drawing.

-No bracket for the "CLR" button.

### Expected Results/How It Should Work:
-Make holes for "OH1A8A1-101 - UFC COMM BEZEL" brackets on the top left (see screenshot 1)

-Cut out the holes for the "OH1A8A1-101 - UFC COMM BEZEL" brackets completely to allow laser cutting (see screenshot 2)

-Modify the button holders to hold the "CLR" button

-(Screenshot 3 is my modification)

### Actual Results/How It Does Work:
-The "CLR" button has no support

-Laser cutting is not possible by following the drawing

### Screenshots:
 [image]
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [x] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH1A8A1-115 - UFC INNER SPACER PLATE

jrsteensen, 2023-05-15:
Already fixed in latest version a couple weeks ago. See #708.

jrsteensen, 2023-05-15:
Good writeup on the issue though!

closed 2023-05-15 by jrsteensen