jrsteensen/OpenHornet #731: Left Console is exploded in top level assy.

jrsteensen, 2023-05-17:
Left console is exploded in top level assembly. Likely a broken joint at the left console level.

Blocking issue for beta1 release.

Exprezz0, 2023-05-18:
Fixed with Left Console v165 and saved with Top Level Assembly v29.  

Additional fixes that I didn't see issues for but needed to be done to fix LC:

_**1.  ASSY, FCS PANEL**_ - few broken joints with the two main problems being the mount for the pro micro and a single hex nut used on the SPDT(SPST?, 100h type).  The hex nut specifically had 2 joints, one that linked it to the origin point and another that joined it to the locking washer (which is what it should have had).  Once I removed the random rigid joint to the origin, I rejoined the hex nut and switch to the panel.  This specific issue caused all of the joints on the Left Console Assembly to have the constraining errors which resolved 90% of the issues. 

I also checked the drawing and updated with the latest version.  I didn't notice any changes in the drawing itself due to the fixing of the joints.  

**_2. ASSY, ANT SEL PANEL_** - The Legend Plate and Light Plate were missing a joint attaching it to the backplate.  Added and saved.

**_3. ASSY, LEFT CONSOLE STRUCTURE_** - Removed duplicate hex nut located towards the back.  Noticed that when I saved it, it got hung up.  It should have saved from v62 to v63, but didn't actually save, so I hid/unhid the origin to make a minor change and resave it. 
 Once I saved it, it was v64 and must have pushed v63 through the pipeline.

**_4. ASSY, LANDING GEAR LEVER PANEL_** - Added joints to the LG LED Mount, LG Lollipop, and the ABSIS PCB assembly (hex standoff screws)

**_5. ASSY, SIM PWR PANEL_** - Added joint to lightplate to fix it to the backplate

**_6. ASSY, THROTTLE QUADRANT_** - Fixed joints for 4x brass inserts & added joint for bracket and spacer

**_7. ASSY, SELECT JETT PANEL_** - Added joint for single SHCS screw connecting backplate & PCB

**_8. *MAJOR* ASSY, LEFT CONSOLE_** - 1) Updated any subassembly links to the newest version based on fix.  2) There were 2x screws and 2x hex nuts used to mount the console interconnect that were floating.

[image]
[image]
[image]

closed 2023-05-18 by Exprezz0

jrsteensen, 2023-05-18:
😃 Thanks @Exprezz0!