jrsteensen/OpenHornet #746: Generate MFG Files: OH2A1-1 - ASSY, LOWER INSTRUMENT PANEL

jrsteensen, 2023-05-18:
Generate the manufacturing files for Generate MFG Files: OH2A1-1 - ASSY, LOWER INSTRUMENT PANEL. 

__Check off each item in issue as you complete it.__

### File generation
- [OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
- [ ] Generate SVG files (if required.)
- [ ] Generate 3MF files (if required.)
- [ ] Generate STEP files (if required.)
- [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)
### Review your files
- [ ] Verify against drawing parts list that all the relevant manufacturing files have been created.
- [ ] Open each SVG in your browser and compare against part to ensure it appears the same and its filename is correct.
- [ ] Open each 3MF in a slicer of your choice and verify geometry matches F360 model and its filename is correct.
- [ ] Open each STEP in a STEP file viewer of your choice and verify geometry matches F360 model and its filename is correct.
### Submit your files
- [ ] Create a github PR against the beta 1 branch with the manufacturing files located in correct location of the release folder.
- [ ] Request a review of the PR.

#### Why a PR?
It gives you credit when I generate the changelog in the release, and (more importantly) adds traceability to the history of the issues.

Arribe, 2023-06-01:
In progress: Arribe

Arribe, 2023-06-01:
Looks like PR #925 didn't automatically close this one.

PR #925 (2023-06-01): Generate MFG Files: OH2A1-1 - ASSY, LOWER INSTRUMENT PANEL#746
# Description

Manufacturing files generated for: Generate MFG Files: OH2A1-1 - ASSY, LOWER INSTRUMENT PANEL

Note: PSU bracket holder names have a V1 in the description, but has been removed to match the drawing and to not cause confusion with two version numbers in the names. (ex: FDM_OH2A1-30_PSU BRACKET LOWER V1_-_V128.3MF)

Close #746

## File generation
[OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
 - [ ] Generate SVG files (if required.)
 - [X] Generate 3M [... cut]
closed 2023-06-01 by jrsteensen