jrsteensen/OpenHornet #771: Generate MFG Files: OH3A3-40 - ASSY, STICK EXTENSION

jrsteensen, 2023-05-18:
Generate the manufacturing files for Generate MFG Files: OH3A3-40 - ASSY, STICK EXTENSION. 

__Check off each item in issue as you complete it.__

### File generation
- [OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
- [ ] Generate SVG files (if required.)
- [ ] Generate 3MF files (if required.)
- [ ] Generate STEP files (if required.)
- [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)
### Review your files
- [ ] Verify against drawing parts list that all the relevant manufacturing files have been created.
- [ ] Open each SVG in your browser and compare against part to ensure it appears the same and its filename is correct.
- [ ] Open each 3MF in a slicer of your choice and verify geometry matches F360 model and its filename is correct.
- [ ] Open each STEP in a STEP file viewer of your choice and verify geometry matches F360 model and its filename is correct.
### Submit your files
- [ ] Create a github PR against the beta 1 branch with the manufacturing files located in correct location of the release folder.
- [ ] Request a review of the PR.

#### Why a PR?
It gives you credit when I generate the changelog in the release, and (more importantly) adds traceability to the history of the issues.

jrsteensen, 2023-05-18:
Blocked pending #725

jrsteensen, 2023-05-20:
Block cleared with #725 resolution.

Arribe, 2023-06-03:
In progress: Arribe

PR #946 (2023-06-03): Generate MFG Files: OH3A3-40 - ASSY, STICK EXTENSION#771
# Description

Manufacturing files generated for: OH3A3-40 - ASSY, STICK EXTENSION

Close #771

## File generation
[OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
 - [X] Generate SVG files (if required.)
 - [X] Generate 3MF files (if required.)
 - [ ] Generate STEP files (if required.)
 - [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)

## Review your files
 - [X] Verify against drawing parts list that all the rele [... cut]
closed 2023-06-03 by jrsteensen