jrsteensen/OpenHornet #800: Generate MFG Files: OH5A1A1-1 - ASSY, RIGHT CONSOLE STRUCTURE

jrsteensen, 2023-05-18:
Generate the manufacturing files for Generate MFG Files: OH5A1A1-1 - ASSY, RIGHT CONSOLE STRUCTURE. 

__Check off each item in issue as you complete it.__

### File generation
- [OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
- [ ] Generate SVG files (if required.)
- [ ] Generate 3MF files (if required.)
- [ ] Generate STEP files (if required.)
- [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)
### Review your files
- [ ] Verify against drawing parts list that all the relevant manufacturing files have been created.
- [ ] Open each SVG in your browser and compare against part to ensure it appears the same and its filename is correct.
- [ ] Open each 3MF in a slicer of your choice and verify geometry matches F360 model and its filename is correct.
- [ ] Open each STEP in a STEP file viewer of your choice and verify geometry matches F360 model and its filename is correct.
### Submit your files
- [ ] Create a github PR against the beta 1 branch with the manufacturing files located in correct location of the release folder.
- [ ] Request a review of the PR.

#### Why a PR?
It gives you credit when I generate the changelog in the release, and (more importantly) adds traceability to the history of the issues.

Arribe, 2023-05-21:
In Progress: Arribe

jrsteensen, 2023-05-21:
Regenerate SVGs of MDF parts with CSKs removed IAW 3.2.d of the HOWTO.

Arribe, 2023-05-21:
Removed CSKs from exported face.  Note there are 3 different screw hole sizes for each of these: wood screws, panhead/round #8 screws, and threaded wood inserts.

PR #838 (2023-05-21): Generate MFG Files: OH5A1A1-1 - ASSY, RIGHT CONSOLE STRUCTURE#800
# Description

Manufacturing files created for OH5A1A1-1 - ASSY, RIGHT CONSOLE STRUCTURE

Close #800

## File generation
[OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
 - [x] Generate SVG files (if required.)
 - [x] Generate 3MF files (if required.)
 - [ ] Generate STEP files (if required.)
 - [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)

## Review your files
 - [x] Verify against drawing parts list th [... cut]
closed 2023-05-21 by jrsteensen