jrsteensen/OpenHornet #816: Generate MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL

jrsteensen, 2023-05-18:
Generate the manufacturing files for Generate MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL. 

__Check off each item in issue as you complete it.__

### File generation
- [OH Wiki HOWTO Link](https://github.com/jrsteensen/OpenHornet/wiki/HOWTO:-Generating-Fusion360-Manufacturing-Files)
- [ ] Generate SVG files (if required.)
- [ ] Generate 3MF files (if required.)
- [ ] Generate STEP files (if required.)
- [ ] Copy the relevant decal PDFs from the art folder to the relevant manufacturing folder (if required.)
### Review your files
- [ ] Verify against drawing parts list that all the relevant manufacturing files have been created.
- [ ] Open each SVG in your browser and compare against part to ensure it appears the same and its filename is correct.
- [ ] Open each 3MF in a slicer of your choice and verify geometry matches F360 model and its filename is correct.
- [ ] Open each STEP in a STEP file viewer of your choice and verify geometry matches F360 model and its filename is correct.
### Submit your files
- [ ] Create a github PR against the beta 1 branch with the manufacturing files located in correct location of the release folder.
- [ ] Request a review of the PR.

#### Why a PR?
It gives you credit when I generate the changelog in the release, and (more importantly) adds traceability to the history of the issues.

Lazarus94, 2023-05-22:
IN PROGRESS

jrsteensen, 2023-05-22:
Missing -14 adapter plate. Please PR it.

PR #843 (2023-05-22): MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL
Generated MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL #816

Close #816
PR #870 (2023-05-24): MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL
Generated MFG Files: OH5A8A1-1 - ASSY, SIM CNTL PANEL

Close #816
closed 2023-05-24 by jrsteensen