jrsteensen/OpenHornet #825: [Bug]: OH5A2A4-1 - Radar Altimeter Discontinued Gear: 615242

Arribe, 2023-05-19:
### Discord Username

Arribe

### Bug Summary

The Servocity gear 615242 is discontinued.  They recommended SKU:  2303-0250-0024 as a replacement, but that gear is too large for a simple drop in replacement.

### Expected Results

Need to identify a replacement gear.

### Actual Results

Cannot buy the spec'd gear, and their suggested replacement is too large.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH5A2A4-1

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.

jrsteensen, 2023-05-19:
Technically, this should be an obsolescence report, not a bug report. 😄

Arribe, 2023-05-19:
too many options...guess I went with the first item I saw on the list.  😁 Printing the gear in PETG is fine.

jrsteensen, 2023-05-20:
Replaced with printed gears and added a #8-32 X .125 set screw in RALT v17

closed 2023-05-20 by jrsteensen