jrsteensen/OpenHornet #845: [Bug]: UIP parts with same part number: OH1A1-28

Arribe, 2023-05-22:
### Discord Username

Arribe

### Bug Summary

On the UIP part number OH1A1-28 is used for both the UPPER UIP FACE MOUNT, RIGHT, and UIP RIGHT GLARESHIELD.

### Expected Results

Parts should have their own part numbers.  If possible please keep UPPER UIP FACE MOUNT, RIGHT associated with OH1A1-28 as that was used to generate the 3mf file.

### Actual Results

Same part number used for 2 different parts.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH1A1-28

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

Discovered during manufacturing file generation when comparing the release drawing to the list of exported files.

jrsteensen, 2023-05-22:
- [ ] Change PN to -29

jrsteensen, 2023-06-03:
Closed with UIP v106

commit 56d278c2 (2023-06-03): Added Flat DXFs of UIP glare shields

Updated right glareshield to -29.
Created DXFs of both UIP glareshields. (SVG export was not possible.)
Closes #845 & #733
closed 2023-06-03 by jrsteensen