jrsteensen/OpenHornet #849: [Bug]: OH3A2A-216 and OH3A2A-218 LOWER PARACHUTE CONTAINER SKIN TRIM ACRL, listed as Acrylic

Arribe, 2023-05-22:
### Discord Username

Arribe

### Bug Summary

OH3A2A-216 LOWER PARACHUTE CONTAINER SKIN TRIM ACRL, listed as Acrylic, but has rounded edges.  Extracted manufacturing parts as FDM printed.

### Expected Results

3D printing would be easier to make for most builders vs cutting acrylic and then rounding the edges with good results.

### Actual Results

Part listed as acrylic on the drawing.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH3A2A-216, OH3A2A-218

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Center Tub (CT)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to PETG
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-04:
Closed with Seat v95

closed 2023-06-04 by jrsteensen