jrsteensen/OpenHornet #850: [Bug]: OH5A2A2-10 WING FOLD HANDLE RETAINING NUT - listed as steel

Arribe, 2023-05-22:
### Discord Username

Arribe

### Bug Summary

OH5A2A2-10 WING FOLD HANDLE RETAINING NUT listed as steel on the drawing.  Can be printed in PETG with good results.

### Expected Results

The nut should be easy to make, and not require sourcing with a machine shop for a one off turning.

### Actual Results

Part is listed as steel on the drawing.

### Applicable Part Numbers

OH5A2A2-10

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to PETG
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Competed in v18

commit 4310b1ee (2023-06-03): Changed -10 of Wing Fold Handle to PETG

Closes #850
closed 2023-06-03 by jrsteensen