jrsteensen/OpenHornet #859: [Bug]: OH4A10-10 TRANSDUCER GAIN REMOTE PLATE - listed as ACRL should be RMRK

Arribe, 2023-05-23:
### Discord Username

Arribe

### Bug Summary

OH4A10-10 TRANSDUCER GAIN REMOTE PLATE - listed as ACRL should be RMRK on drawing

### Expected Results

Should be RMRK for engraving the lettering.

### Actual Results

Part is listed as acrylic.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH4A10-10

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to RWMK
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Fixed in v7

closed 2023-06-03 by jrsteensen