jrsteensen/OpenHornet #862: [Bug]: OH4A2A1-200 - ASSY, LDG GEAR LVR MECH_-_v4.pdf sheet 3 and 4 dupes

Arribe, 2023-05-23:
### Discord Username

Arribe

### Bug Summary

Page 3 and 4 are the same except for a small offset in part locations.

### Expected Results

Duplicate views of the same parts not needed.

### Actual Results

Page 3 and 4 are the same

### Applicable Part Numbers

OH4A2A1-200

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Delete sheet 4
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Fixed in DWG v5

closed 2023-06-03 by jrsteensen