jrsteensen/OpenHornet #879: [Bug]: Missing Art File - OH4A8A3-11 - NUC WEAP PLACARD

Arribe, 2023-05-23:
### Discord Username

Arribe

### Bug Summary

Working on completing the manufacturing files and didn't find the nuc weapon placard in the art folder.  Part is listed on the design as RMRK, but only has a .005 thickness.

### Expected Results

expected to find a pdf to copy to the manufacturing folder.

### Actual Results

no artwork

### Applicable Part Numbers

OH4A8A3-11

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to RWMK
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
This will be generated as a placard in future beta.