jrsteensen/OpenHornet #880: [Bug]: OH4A1-11 USB COVER PLATE listed as steel, vs acrylic

Arribe, 2023-05-23:
### Discord Username

Arribe

### Bug Summary

Left console USB cover plate listed as steel in the design.  Should be acrylic for ease of building and consistency with other backplates.

### Expected Results

Acrylic

### Actual Results

steel

### Applicable Part Numbers

OH4A1-11

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to ACRL
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Fixed in LC v166

closed 2023-06-03 by jrsteensen