jrsteensen/OpenHornet #887: [Bug]: OH4A3A3-10 1.5 IN CASE design material listed as ABS

Arribe, 2023-05-24:
### Discord Username

Arribe

### Bug Summary

OH4A3A3-10 1.5 IN CASE design material listed as ABS

### Expected Results

Part can be printed in PETG

### Actual Results

Part doesn't need to be printed in ABS

### Applicable Part Numbers

OH4A3A3-10

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change to PETG
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Fixed in drawing v7

closed 2023-06-03 by jrsteensen