jrsteensen/OpenHornet #890: [Bug]: Throttle Assembly Material Updates

Arribe, 2023-05-24:
### Discord Username

Arribe

### Bug Summary

While exporting parts here are some suggestions for reviewing the material listed:

OH4A9-19 - AFTERBURNER DETENT, OH4A9-15 - IDLE DETENT both listed as aluminum, may be challenging to make could be printed.

OH4A9-10 LH THROTTLE BASE PLATE, OH4A9-11 RH THROTTLE BASE PLATE both listed as "SS" - should be ALU?

OH4A9-118 COMM 4-WAY SWITCH HAT listed as PETG, looks good as RESN.

OH4A9-120, OH4A9-114, OH4A9-113 TDC switch parts listed as PETG, prints well as RESN.

### Expected Results

Suggested material review for drawing listed in summary

### Actual Results

Some parts may be better with updated materials per my initial beta build.

### Applicable Part Numbers

multiple

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Throttle

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Change -15 to PETG
- [ ] Change -10/-11 to ALU
- [ ] Change -118, -120, -114, -113 to RESN
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Closed with v46

closed 2023-06-03 by jrsteensen