jrsteensen/OpenHornet #891: [Bug]: OH4A7A1-13 COMM PANEL LEGEND PLATE - 8's do not export correctly

Arribe, 2023-05-24:
### Discord Username

Arribe

### Bug Summary

Need to increase depth of 8's on COMM PANEL LEGEND PLATE by .003 to get them to properly extract as SVGs.

### Expected Results

The full 8 digit should be extracted to the legend plate.

### Actual Results

Only the top half of the 8 extracts correctly.

### Applicable Part Numbers

OH4A7A1-13

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-05-25:
- [ ] Add required .003 extrusion.
- [ ] Generate new drawing and commit PDF to release folder.

jrsteensen, 2023-06-03:
Closed in v21

closed 2023-06-03 by jrsteensen