jrsteensen/OpenHornet #892: [Bug]: OH4A3A3-12 BRAKE PRESSURE INDICATOR SHROUD - ACRL needs deeper extrusion

Arribe, 2023-05-24:
### Discord Username

Arribe

### Bug Summary

The ACRL face doesn't extract as SVG completely for the characters: P and 0

They need to be deepened by .003.

### Expected Results

P and 0's are extracted correctly as SVG.

### Actual Results

the inner portion of the characters are not extracted correctly.

### Applicable Part Numbers

OH4A3A3-12

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2023-05-24:
I took a look at this and this appears fine to me
OH4A3A3-12 the Shroud
[image]

I ran shaper on it and it looks correct,

The files in the releases also look correct,

Can you clarify the issue please as I do not see an issue with the P and O's?

Arribe, 2023-05-25:
When I exported the P and 0 were solid white with no internal black circles. I lowered by .003 to get them out. But if it’s just me then that’s fine. The files I exported are worked by temporarily lowering them. Don’t need to do it again.

jrsteensen, 2023-05-25:
I'd prefer if there are issues from the model to get them fixed now so we don't inadvertently generate a bad SVG in the future.

jrsteensen, 2023-06-03:
Fixed.

closed 2023-06-03 by jrsteensen