jrsteensen/OpenHornet #896: Missing "Lip or Rings" on bottom through holes on the Radar Alt Case (all three)

Krikeee, 2023-05-25:
### Discord Username

Krikee

### Bug Summary

The holes used to fasten the POT Mount (OH5A2A4-21) and the Radio Alt Stand off (OH5A2A4-15) parts to the Radio Alt Case seem to be missing the inner "lip or ring" on the bottom side of the Case. These rings can be seen in the holes in the side of the case. The issue being the screws simply pass through the holes on the bottom and do not retain the attached parts.
[image]

### Expected Results

Through holes on bottom should have a lip to allow the screw to support the attached part

### Actual Results

Through holes on the bottom allow the screw to pass through the case and do not support the attached parts. Specifically the POT Mount and the Radio Alt Stand Off part.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH5A2A4-16 - RALT CASE; OH5A2A4-21 - RALT POT MOUNT; OH5A2A4-15 - RALT STEPPER STANDOFF

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-06-04:
Hmmm, for some reason, the counterbore (CBORE) holes became thru holes. Looking at this now.

jrsteensen, 2023-06-04:
Fixed in v20, and new mfg files exported for the case.

closed 2023-06-04 by jrsteensen