jrsteensen/OpenHornet #903: [Bug]: OH2A2A1A1-16 - JETT STATION SEL INDICATOR LEGEND DIFFUSER - listed as resin

Arribe, 2023-05-30:
### Discord Username

Arribe

### Bug Summary

Drawing / model for OH2A2A1A1-16 - JETT STATION SEL INDICATOR LEGEND DIFFUSER - listed as resin vs acrylic

### Expected Results

expected part to by acrylic

### Actual Results

listed as resin

### Applicable Part Numbers

OH2A2A1A1-16

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Arribe, 2023-05-30:
Should be listed as ABS to match the other -23 diffuser.

jrsteensen, 2023-06-04:
Set all diffusers to acrylic in JETT SELECT v5.

closed 2023-06-04 by jrsteensen