jrsteensen/OpenHornet #947: [Bug]: OH1A4-1 - ASSY, EYEBROW WARNING MODULE, LEFT duplicate diffuser parts, and material update

Arribe, 2023-06-05:
### Discord Username

Arribe

### Bug Summary

OH1A5-18 appears to be the same part as these copies for just the left module: OH1A4-24 EYEBROW WARNING SMALL LEGEND DIFFUSER (1) and OH1A4-25 EYEBROW WARNING SMALL LEGEND DIFFUSER (2).  Could streamline by removing the duplicates and setting it as another instance of OH1A5-18

OH1A4-26 EYEBROW WARNING LARGE LEGEND, MASTER CAUTION listed as RMRK.  But all the other legends are listed as ACRL.

### Expected Results

If -24 and -25 are the same as OH1A5-18 then just use another instance of -18.

legend part material consistency

### Actual Results

extra part numbers

one legend out of the entire module listed as RMRK vs ACRL.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH1A4-1

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-06-26:
- [ ] Change one RWMK diffuser to acrylic.

DIffusers should stay different part numbers because they are different colors.

jrsteensen, 2023-06-30:
1. -26 changed to ACRL in model
2. Updated Drawing
3. -26 MFG Filename was correct

Closed in 1a25e7c91e18bed964c930845b6a67ddff20a545

commit 1a25e7c9 (2023-06-30): Updated L EWM Drawing

Closes #947
closed 2023-06-30 by jrsteensen