jrsteensen/OpenHornet #951: OH5A8A1-12_SIM CNTL LEGEND PLATE: Engraving for the large multi-position switch should not include the hole

icemarkom, 2023-06-22:
### Discord Username

icemarkom

### Bug Summary

This is Lightburn preview of the engraving and cutting of the "OH5A8A1-12_SIM CNTL LEGEND PLATE" from beta1 release branch.

[image]

There should be a gray circle limiting the engraving path to not include the part that will be cut out.

### Expected Results

There should be a gray circle limiting the engraving path to not include the part that will be cut out.

### Actual Results

Engraving the part that will be discarded happens.

### Applicable Part Numbers

OH5A8A1-12_SIM CNTL LEGEND PLATE

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

icemarkom, 2023-06-22:
This would be my expected output from cut/engraving pass:

[image]

kbastronomics, 2023-06-22:
The circle exists,  it's how lightburn is importing this feature,  it brings it in as a cut layer which is correct as the center should be cut out,  and is the same for all the knobs with the large backlight engraving
Shaper does not export any addition layers which would would be needed to make it work so the modification would have to be made in lightburn or other engraving software.

So what needs to happen in light burn is the duplicate the cut layer as an engrave layer and leave both in same place do you get both the cut and keep out area for the engrave.

in my case I ignore it and let it grave,  its costing no additional time to do so

jrsteensen, 2023-06-22:
No issue for manufacturing files, per @kbastronomics. Will not be acted on at this time. But thank you for the bug report! I always appreciate them!

closed 2023-06-22 by jrsteensen