jrsteensen/OpenHornet #954: [Bug]:  Cover Glass SVG file missing from Manufacturing FIles for Standby Instrument Module

kbastronomics, 2023-06-27:
### Discord Username

smgvbest (aka Sandra)

### Bug Summary

The Cover Glass used in 3 of the instruments in the Standby Instrument Module is missing.   This part would be ACRL_.062_OHC0007-11 - 2IN GAUGE COVER GLASS_-_V1.SVG

### Expected Results

File should be created so part can be made by users

### Actual Results

generated part for self use and filled bug report

### Applicable Part Numbers

OHC0007-11 - 2IN GAUGE COVER GLASS

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Arribe, 2023-06-27:
OHC is a common part.  Did you look in folder? .../release\OHC_Common_Parts\OHC0007 - INSTRUMENT COMPONENTS\Manufacturing

jrsteensen, 2023-06-30:
Part exists in appropriate common folder. Unfounded bug.

[image]

closed 2023-06-30 by jrsteensen