jrsteensen/OpenHornet #969: [Bug]: Ejection Seat Misssing details on the Ejection Handle,  SEAT ARMING LEVER and MANUAL OVERRIDE HANDLE

kbastronomics, 2023-07-05:
### Discord Username

smgvbest (aka Sandra)

### Bug Summary

it looks like a few of the parts on the ejection seat lost thier detailes (legends and/or caution stripes.
OH3A2A-133 - SEAT ARMING LEVER
OH3A2A-132 - MANUAL OVERRIDE HANDLE
OH3A2A-1300 - ASSY, EJECTION HANDLE

### Expected Results

The parts should have all the correct labeling and caution striping

### Actual Results

The should have but they don't 

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH3A2A-133 - SEAT ARMING LEVER OH3A2A-132 - MANUAL OVERRIDE HANDLE 

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Center Tub (CT)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-09-09:
- [ ] 1. Fusion Model Updates
- [ ] 2. Drawings Revised with detail/finish views of each
- [ ] 3. New PDF drawings and manufacturing files generated

kbastronomics, 2024-09-09:
what i s best method to update he Seat Arming Handle,  I've done it on my local copy.
can I export and import or do you prefer to edit the OH model and do the same steps to it that I did on the local copy.

jrsteensen, 2024-09-09:
Best (and least painful for drawing updates) is to natively redo it to the existing model.