jrsteensen/OpenHornet #972: Interconnect: Brk Press Gauge LED wrong current in interconnect

Higgins-OH, 2023-08-01:
### Summary:
The LED for the Brake Pressure Gauge is a Neopixel 1938 with 5V.
According to the interconnect (draft 20221201) it is connected to the 12V.

### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software

jrsteensen, 2023-08-06:
Draft 20221201 is massively outdated. Use interconnect drawing in release folder of beta1 branch.

Issue still valid on latest.

jrsteensen, 2023-08-06:
Fixed in 29a55098f0a2560070b0ed115b2a45cc64700cf9.

To be uploaded once all interconnect issues are resolved.

commit b2114a54 (2023-08-06): Fixed BRK PRESS GAUGE LED Issues

Fixed #972 & #976
closed 2023-08-06 by jrsteensen