jrsteensen/OpenHornet #973: interconnect: Wing Fold microswitches probably wrong name

Higgins-OH, 2023-08-01:
### Summary:
interconnect draft 20221201:
The microswitches for the Wing Fold are named
- Wing Fold Unlock
- WNG Fold Hold
- WNG Fld Spread

Instead of "Hold" (the middle position of the handle), it should be named "Fold", so both rotation microswitches are used.

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software

Arribe, 2023-08-01:
[image]

jrsteensen, 2023-08-06:
Fixed in 7182c503bad53378d8f6cd591725b93fc0b9cb15

commit 7182c503 (2023-08-06): Fixed Wing Fold SW Position Name

SW Name was: WNG FOLD HOLD; Is: WNG FOLD FOLD
Net label was: WF_HOLD; Is: WF_FOLD
Closes #973
closed 2023-08-06 by jrsteensen