jrsteensen/OpenHornet #975: [Bug]: Alt Radar - not enough room between pot and PCB for standard Dupont connector

Higgins-OH, 2023-08-02:
### Discord Username

Higgins

### Bug Summary

There isn't enough room between the pot and the PCB of the Radar Alt to fit a Dupont connector in.
0.5 in between pot and the outer margin of the case (= the PCB).
A Dupont connector plus the spacers of the pins on the PCB are 0.6 in.
Making the case a bit longer could help with that issue. Alternatively: A PCB redesign with the pins somewhere else.

### Expected Results

The PCB is supposed to fit on the case with all the connectors on it.

### Actual Results

The PCB can't be screwed on to the case with everything assembled.

### Screenshots/Images/Files

[image]
[image]

### Applicable Part Numbers

OH5A2A4-16 RALT CASE

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2023-08-02:
could we use a right angle dupont instead?

Higgins-OH, 2023-08-02:
Horizontal pins could help. I'll try them.

Arribe, 2023-08-02:
An interim solution may be using a set of standoffs to move the PCB down off the housing to create space for the connectors.

Krikeee, 2023-08-03:
I have not checked clearances but could just extending the "case" a 1/2 inch or so be a solution?

kbastronomics, 2023-08-04:
I see 4 option in order of increasing complexity/cost

1. Drop the dupont and solder wires direct to pcb or header
2. Use a right angle header (could have space issue though,  needs more investigation)
3. Add standoffs for the PCB adding the needed space 
4. extend the pcb (see attached for a zipped .3mf extending it by 0.6" for someone wanting to try it)

#1 is the simplest solution,  no cost, no reprinting parts, no getting parts.  just make the wires long enough for a service loop

[OH5A2A4-16 - RALT CASE.zip](https://github.com/jrsteensen/OpenHornet/files/12259040/OH5A2A4-16.-.RALT.CASE.zip)

Krikeee, 2023-08-04:
I posted my RAlt modification in my build log if anyone would find it helpful. I extended the case length by 1.2 inches. Unless there is something externally that would get in the way, it seems like a very simple solution.

jrsteensen, 2023-08-19:
Extended case 1.5" to also clearance USB connector for RALT from the vent in NHA.
Closed with d007555c54b83df048819743410a86804034fa2d & RALT v22.

commit d007555c (2023-08-19): Updated RALT Case to be 1.5" longer

Clearances internal connectors and moves USB connection to not interfere with vent in R CONSOLE
Closes #975
closed 2023-08-19 by jrsteensen