jrsteensen/OpenHornet #977: [Bug]: Startech USB32HD4 updated, size changed

Higgins-OH, 2023-08-17:
### Discord Username

Higgins

### Bug Summary

The USB32HD4 is no longer available in the size/shape from the original design. The new length is now 4 inches.
See data sheet: https://media.startech.com/cms/pdfs/usb32hd4_datasheet.pdf

### Expected Results

The bracket needs to be a bit longer for the adapter to fit.

### Actual Results

The bracket is too short.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH2A1-29

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-01:
- [ ] Update Model
- [ ] Evaluate placement in LIP
- [ ] Update LIP
- [ ] Update Manufacturing File for Bracket

jrsteensen, 2023-11-04:
Closed with b39025a273f53cfa161401ad7a174843c32aa3e3

commit b39025a2 (2023-11-04): Updated USB TO HDMI Adapter

Updated Adapter Mount for correct adapter dimensions.
Closes #977
closed 2023-11-04 by jrsteensen