jrsteensen/OpenHornet #978: [Bug]: Model design issue - OH1A6-10 - SPIN PANEL BACKPLATE:1

CopperLu-0314, 2023-08-24:
### Discord Username

cooperlu0314

### Bug Summary

The backplate can't fit properly because of the LED pins and resistor on the top layer of the backlight PCB.

### Expected Results

The backplate and PCB should fit together without any gap.

### Actual Results

Because of the LED pins and resister, there's going to be a gap between two parts.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH1A6-10

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-01:
- [ ] Add hole to OH1A6-10
- [ ] Update manufacturing File

jrsteensen, 2023-11-03:
Bumping to beta2 - relocate LED & Resistor to open area where cutting a hole thru backplate will not be visible.

[image]

- [ ] Update SPIN RCV Panel PCB
- [ ] Add hole to backplate
- [ ] Update manufacturing files for both PCB and backplate.