jrsteensen/OpenHornet #981: OH2A1A1-43 - LOWER RIGHT LIP AUX SUPPORT BRACKET - Material in PDF listed as MDF, should be PETG

5k33tz, 2023-08-30:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
In PDF "OH2A1A1-1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE_-_v6.pdf" under v1.0.0-beta1, the material for PN 34 "LOWER RIGHT LIP AUX SUPPORT BRACKET" is listed as MDF, but part material should be PETG.

### Expected Results/How It Should Work:
(Explain how you think it should work here. Be very specific.)

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
[image]
<br>**PRO-TIP:** Just paste the images here.*

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [X] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2023-11-01:
- [ ] Update material of OH2A1A1-43 to PETG
- [ ] Update OH2A1A1-1 (LIP Structure) drawing.

jrsteensen, 2023-11-03:
Closed in a89bd74b070fb418a1f26b972df3f3a034d4d53c

closed 2023-11-03 by jrsteensen