jrsteensen/OpenHornet #982: [Bug]: Interconnect Diagram: Analog Stick Breakout listed as +5v but only gets 3.3v from controller

Arribe, 2023-09-29:
### Discord Username

Arribe

### Bug Summary

In reviewing the interconnect while reviewing components and wiring, noted that the throttle controller sends 3.3v to the inner grip PCB.  The inner grip's Analog Stick Breakout connecter is listed as 5v.

### Expected Results

Since 3.3v is only available to the analog stick connector / PCB, shouldn't the voltage be listed ad 3.3v and not 5v?

### Actual Results

listed as 5v.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH Interconnect Diagram

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Throttle

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

kbastronomics, 2023-09-29:
5V is thier maximum voltage per the datasheet, Since these are just pots so 3.3V is fine.

jrsteensen, 2023-11-01:
Upon second thought, the PCB CAN take UP TO +5V, so not sure its wrong to call it 5V....going to close this one for now.

closed 2023-11-01 by jrsteensen