jrsteensen/OpenHornet #983: [Bug]: ECAD - Throttle Inner Grip Arduino Pro-Mini Description Update

Arribe, 2023-09-29:
### Discord Username

Arribe

### Bug Summary

The throttle inner grip sketch, BOM or other files doesn't list the type of ProMini needed.  It should clarify voltage selection, and if available a part number from one of the typical suppliers would helpful.

### Expected Results

There should be enough information in the sketch / BOM or other file to help builders order the correct components, without opening the KiCAD files to reverse engineer the intended Arduino pro-mini should be the 3.3v version.  Screen shot of sketch shows a potential option that is similar to what is on the ABSIS ALE sketch to specify the correct pro-micro.

### Actual Results

The files do not list the pro-mini version to buy.  Users (like me), may order the 5v version thinking that it should match the others.  Also the throttle controller takes an external 5v connection, but one would need to study the KiCAD PCB file to know that only 3.3v is passed ot the inner throttle grip PCB.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

ECAD\PCBs\Functional PCBs\HOTAS_Throttle Inner Grip

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Throttle

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-02:
- [ ] Add note to schematic specifying 3.3V Pro Mini.

jrsteensen, 2023-11-03:
Closed with latest commit.

commit 299d8313 (2023-11-03): Updated HOTAS_Throttle Inner Grip schematic

Added 3.3V ProMini note to schematic.

Closes #983
closed 2023-11-03 by jrsteensen