jrsteensen/OpenHornet #984: [Bug]: Cutout for SIM PWR PCB missing on spacers

Higgins-OH, 2023-10-01:
### Discord Username

Higgins

### Bug Summary

Left Aux Spacer and A-B Left Outboard Spacer need a cutout for the SIM PWR PCB. 

### Expected Results

The PCB should be sunk into the spacers.

### Actual Results

At the moment the SIM PWR panel can't sit flush with the spacers because of the PCB.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH4A1A1-18, OH4A1A1-19, 

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Left Console (LCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-02:
- [ ] Update LCON Structure and OH4A1A1-18, OH4A1A1-19 to provide relief for PCB.
- [ ] Update appropriate manufacturing files.

jrsteensen, 2023-11-03:
Closed with b836af9690b2af760b4c977156209c86332f54ee

closed 2023-11-03 by jrsteensen