jrsteensen/OpenHornet #985: [Bug]: RAlt Controller PCB: BL Silkscreen missing

Higgins-OH, 2023-10-15:
### Discord Username

Higgins

### Bug Summary

On the RAlt Controller PCB the labels "in" and "out" are missing.

### Expected Results

It would be better to see which connector is for "in" and which is for "out".

### Actual Results

The BL silkscreen labels are missing.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

CONTROLLER_Radar Altimeter

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

### Miscellaneous Info

There's no version number on the PCB. That would be helpful, too.

jrsteensen, 2023-11-02:
- Low priority beta1 update, may be bumped to beta 2.

- [ ] Update PCB layout & Gerbers.

jrsteensen, 2023-11-03:
Closed with f9735d513a1dfb10959a278252fd3cc90c631ef7

commit f9735d51 (2023-11-03): Updated CONTROLLER_Radar Altimeter PCB

Made BL IN/BL OUT connector silkscreens visible.
Updated BOM.

Closes #985
closed 2023-11-03 by jrsteensen