jrsteensen/OpenHornet #987: [Bug]: HUD Panel PCB: pigtails for BL needed

Higgins-OH, 2023-10-17:
### Discord Username

Higgins

### Bug Summary

The left DDI display can't get close to the UIP front plate due to the HUD Panel BL connectors.
We need pigtails (in the plans and mentioned in the interconnect), so the display can touch the front plate again.

### Expected Results

The display needs to be able to touch the front plate.

### Actual Results

The connectors are too big, there is a gap between the front plate and the display.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH1A7-101

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Upper Instrument Panel (UIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-02:
- [ ] Move connectors away from PCB like other pigtails in the model.
- [ ] Note pigtail in interconnect.

jrsteensen, 2023-11-03:
Completed.

closed 2023-11-03 by jrsteensen