jrsteensen/OpenHornet #988: [Bug]: Interconnect: Defog Panel missing in the BL interconnect (v4)

Higgins-OH, 2023-10-17:
### Discord Username

Higgins

### Bug Summary

The Defog Panel is missing in the BL interconnect.

### Expected Results

It needs to be included.

### Actual Results

It's missing.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH5A5A2-13

### Release Version

1.0.0-beta.1

### Category

Electrical (PCBs/Wiring/etc)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-02:
- [ ] Its missing. It should be added to the interconnect.

jrsteensen, 2023-11-03:
Completed.

closed 2023-11-03 by jrsteensen