jrsteensen/OpenHornet #989: [Bug]: LIP to RC bulkhead panel mount USB cable too short

Higgins-OH, 2023-10-26:
### Discord Username

Higgins

### Bug Summary

The panel mounted USB cable from the LIP to the RC bulkhead panel is to be 3ft long. That is too short to reach the USB hub on the left side of the LIP.

### Expected Results

Longer USB cable.

### Actual Results

Too short. Can't reach the hub.

### Applicable Part Numbers

1698-03C at OHC0013-4

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-10-26:
PN should be 1698-0**4**C

- [ ] Update OHC0013-4 model/drawing
- [ ] Update interconnect references

jrsteensen, 2023-11-03:
Closed with ef566f58115c1d265820e8b34f6637ce78f00e53

commit ef566f58 (2023-11-03): Changed R LIP Bulkhead USB cable to 4' L

Closes #989
closed 2023-11-03 by jrsteensen