jrsteensen/OpenHornet #991: [bug]: Power circuits need diodes for power backfeeding prevention

Higgins-OH, 2023-11-02:
- 5V/12V acc 
- if possible on ALEs

### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [x] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software

jrsteensen, 2023-11-03:
- [ ] Add Shottkey Diodes as reverse polarity protection to all circuits on Power Distro PCB
- [ ] Add Shottkey diode as reverse polarity protection to all power lines on ALE

Higgins-OH, 2023-11-03:
Some kind of zener diode (or whatever prevents the connection of 5V and 12V through the ProMicros) on the 12V circuit of future ALEs wouldn't be bad, too. This way the 12V devices won't draw current through the ProMicro from the 5V circuit. In case 12V isn't provided, only 5V through USB, that could prevent overloading the system.

jrsteensen, 2023-11-05:
ALE Updates closed with 3de2140d16f4f755b07b172f658727227580eda7

commit 3de2140d (2023-11-05): Updated ABSIS ALE power circuit.

- Added 1N5819WS (D1) between 12V SUPPLY and RAW pin of Pro-Micro.
- Incremented to Rev 3.
- Added better verbiage on PCB of how to terminate RS485 bus.
- Added note to schematic to remove F1/ensure JP1 is not soldered on Pro-Micro if external power is applied.

Closes #991
closed 2023-11-05 by jrsteensen