jrsteensen/OpenHornet #1007: SVG File for Standby Alt looks to be missing the 3 x "0" s on the face [Bug]:

Ben-F111, 2023-11-21:
### Discord Username

Ben-F111

### Bug Summary

THe Downloaded SVG file for the DIal Face for the Standby looks to be missing the 3 x "0"s

[image]

### Expected Results

The face should have 3 "0"s next to the 2 cut outs 
[image]

### Actual Results

File needs an update 

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

RMRK_.063_OH2A7A1-204_ALTIMETER DIAL FACE_-_V13

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-11-21:
For some reason, the tool in fusion wont generate those three zeros. Odd. May have to smack them in with a vector art program.

Ben-F111, 2023-11-21:
Cheers thanks.

kbastronomics, 2023-11-21:
load the file into your software,  Lightburn in my case and it's fine
only the preview is in error

[image]

once loaded into lightburn
[image]

Ben-F111, 2023-11-21:
Got the same view on Visio without the O’s

kbastronomics, 2023-11-21:
its the way visio and the preview render 

this is from visio,   I ungrouped and move the layer

[image]

jrsteensen, 2023-11-22:
Its there, just not colored. Limitation of software visibility until moved into program. Still manufacturable, just not ideal, but no ability to fix at this time.

closed 2023-11-22 by jrsteensen