jrsteensen/OpenHornet #1008: bug: DDI/AMPCD controller PCB: CPL with DIP socket

Higgins-OH, 2023-11-21:
### Summary:
The DIP-24 socket is in BOM and CPL, although the through-hole parts are supposed to be soldered manually.

### Expected Results/How It Should Work:
No DIP in CPL.

### Category:
- [x] ECAD (PCB Design or other electrical hardware)
### Type:
*Check one item.*
- [x] Bug
### Applicable End Item:
*Check one item.*
- [x] Top Level Assembly          

### Associated Filename(s):
OHC0015-10

Arribe, 2023-11-21:
Through hole parts are an option to have JLCPCB install, but they have generally been cheaper just to do manually.  The last ALE order I placed I had JLC install the Molex connectors because it was way cheaper than buying the connectors from Digikey or Mouser and doing it myself.  The other through hole parts weren't cost effective, or in stock.  Having the through hole parts in the BOM / CPL at least gives the option to have them installed, if one doesn't want them installed then one can choose the don't place part option.

Higgins-OH, 2023-11-21:
Sounds reasonable, but in this case all the through-hole parts should be in the CPLs. ALE was the first and only I noticed so far.

kbastronomics, 2023-11-21:
Totally agree. 
Have them in the bom and cpl files

jrsteensen, 2024-03-01:
Part of #719 for functional PCBs.

closed 2024-03-01 by jrsteensen