jrsteensen/OpenHornet #1010: [FEATURE REQUEST]: OH2A2A1-1 - ASSY, LEFT LIP MODULE_-_v5.pdf: Nr. 5 missing

Higgins-OH, 2023-12-08:
### Discord Username

Higgins

### Feature/Enhancement Summary

Nr. 5 in the drawings is missing.

### Justification

It helps builders to easily recognize the proper screws.

### Applicable Part Numbers

OH2A2A1-1 - ASSY, LEFT LIP MODULE_-_v5.pdf

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Lower Instrument Panel (LIP)

jrsteensen, 2023-12-08:
Update as noted. Drawing change only - no change to form, fit or function.

jrsteensen, 2024-02-25:
Closed with d5fc02c1033c68180c67b0ee55a0fd810586db2b

closed 2024-02-25 by jrsteensen