jrsteensen/OpenHornet #1012: [Bug]: OH3A2-1 - ASSY, EJECTION SEAT, SJU-17 - Part Length Mismatch from McMaster

5k33tz, 2023-12-16:
### Discord Username

5k33tz

### Bug Summary

In PDF "OH3A2-1 - ASSY, EJECTION SEAT, SJU-17_-_v27.pdf" under v1.0.0-beta1, the length for PN 36 "PIN, DOWEL, .125 OD X .825L, ALLOY STEEL" is listed as having a length of .825, but the actual PN from McMaster ("98381A474") has a length of .875 (7/8ths).

### Expected Results

Submitting bug report to change value in PDF from .825L to .875L.

### Actual Results

Submitting bug report to change value in PDF from .825L to .875L.

### Screenshots/Images/Files

[image]
[image]

### Applicable Part Numbers

OH3A2-300

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Center Tub (CT)

### Built to print?

- [ ] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2023-12-16:
Change COTS Part Description from .825L to .875L and update drawing.

jrsteensen, 2024-02-26:
Closed with 3817d9ea9e37331535bfc3bfa06bb69765b1578a

commit 3817d9ea (2024-02-26): Corrected typo in 98381A474 description in -300

Corrected typo in 98381A474 description in -300 from .825L to .875L.
Closed #1012
closed 2024-02-26 by jrsteensen