jrsteensen/OpenHornet #1013: [Bug]: Manufacturing Files - Mistakes in the name of some Acrylic plates

Thibaud-54, 2023-12-21:
### Discord Username

Thib-O #5205

### Bug Summary

Here is some plates 0.25 thick (according to the file name) which actually could be 0.188 as usual. Double checked with Fusion for OH5A6A1-11 INTR LT LIGHTPLATE. I made a Excel table which summarize panel with 0.25 thickness information in the file name. It mainly concerns the Right console.

[image]

### Expected Results

Double check/Correct the thickness in the name of the files. 

### Actual Results

The thickness in the file name does not correspond to the real thickness in Fusion.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH5 - Light plates panels

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Right Console (RCON)

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

Arribe, 2023-12-21:
[image]

Arribe, 2023-12-21:
In progress

jrsteensen, 2023-12-29:
Closed with #1014

closed 2023-12-29 by jrsteensen