jrsteensen/OpenHornet #1018: [Bug]: Drawing: OH3A3-100 - ASSY, STICK YOKE_-_v17.pdf missing part

Arribe, 2024-01-17:
### Discord Username

Arribe

### Bug Summary

Drawing doesn't list 608-2RS bearing as a part, nor does it show it on sheet 10 or 11 where it would be installed.

### Expected Results

The drawing should show the bearings, and list them as a part of the OH3A3-119 assembly.

### Actual Results

If one were to build the stick according to the drawings they wouldn't know to buy the 608-2RS bearings, and can overlook where they are installed.

### Screenshots/Images/Files

[image]

### Applicable Part Numbers

OH3A3-100 - ASSY, STICK YOKE

### Release Version

1.0.0-beta.1

### Category

Mechanical (Structure/Panels/Mechanisms)

### Applicable End Item(s)

Flight Stick

### Built to print?

- [X] I built (or attempted to build) the part to the OpenHornet print without any deviations.
- [ ] I am not building this part to the OH print. (List deviations in detail in the Miscellaneous Info text area below.)

jrsteensen, 2024-02-13:
FN 1 should have its own drawing. Will update.

jrsteensen, 2024-02-25:
Closed with 9b48ed09b8e85744a749f0d2b87c1807d7ed7207

commit 9b48ed09 (2024-02-25): Added sheet for -119.

Closes #1018
closed 2024-02-25 by jrsteensen